betaflight/blackbox-log-viewer #473: Not able to open black box log

SinnedFpv, 2020-09-30:
Hello everyone,

I am having some hard times opening my logs for the black box. Every single time i try to open it i have the following error message: 

Error opening log: No log in this file could be parsed successfully

I am running 
clracing f7 v2.1 with mpu6000 gyro
BF 4.2
Blackbox is set to gyro_scaled 1khz serial port
Serial port is Set on uart1 under peripherals 
Black box viewer is on 3.5

I have been digging a lot to solve the issue but all i can find is old post frok bf 3.2 and before and issues have never been solved

Anyone has any idea what is going on??

Thank you for the help

McGiverGim, 2020-09-30:
If you attach a log maybe we can help you.

SinnedFpv, 2020-09-30:
https://drive.google.com/drive/folders/1PlnpaXKuhoz6ZhbGR2IcR6t9O3yBvk3d?usp=sharing

This is the link to the log on my FC
I will paste my dump file if it it can help. I have all the log file for each flight I have done but when I try to open it I have the error message. 
 

# dump

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF7, version: b69c3a8d, date: 2019-10-19T22:37:25Z

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# name: Sinned

# resources
resource BEEPER 1 B04
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 A01
resource MOTOR 6 C08
resource MOTOR 7 C09
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
[... 752 lines cut ...]
set osd_ah_max_pit = 20
set osd_ah_max_rol = 40
set osd_ah_invert = OFF
set osd_logo_on_arming = OFF
set osd_logo_on_arming_duration = 5
set osd_tim1 = 2560
set osd_tim2 = 2561
set osd_vbat_pos = 234
set osd_rssi_pos = 2518
set osd_link_quality_pos = 2486
set osd_rssi_dbm_pos = 185
set osd_tim_1_pos = 234
set osd_tim_2_pos = 2134
set osd_remaining_time_estimate_pos = 234
set osd_flymode_pos = 234
set osd_anti_gravity_pos = 234
set osd_g_force_pos = 234
set osd_throttle_pos = 234
set osd_vtx_channel_pos = 2113
set osd_crosshairs_pos = 205
set osd_ah_sbar_pos = 206
set osd_ah_pos = 78
set osd_current_pos = 2497
set osd_mah_drawn_pos = 234
set osd_motor_diag_pos = 2

McGiverGim, 2020-10-01:
The logs don't record the data. They contain the header of the Blackbox, but not the data. For this reason the blackbox can't open it.

So the problem is not a the Blackbox app, is more a firmware/configuration problem.

- If you think is a Configuration problem, use rcgroups or the official Betaflight Slack chat/facebook group asking for help.
- If you think is a firmware bug, open an issue in the firmware github: https://github.com/betaflight/betaflight/issues read the template that appears and fill all the info asked. 

One suggestion: don't attach a `dump` to any issue. It gives too much information, impossible to see what settings you have modified. A `diff` is always enough and more useful.

SinnedFpv, 2020-10-01:
ok, perfect thank you I will open a new ticket in the appropriate git hub

closed 2020-12-19 by stale[bot]