betaflight/blackbox-log-viewer #486: android versiOn of blackbox

butterSnake, 2020-12-14:
Have got an android tablet so I don't have to Haul the full laptop to the field 
An android version of this would be super rad as the paid racedrone version won't run logs from flash chips so it's totally useless

b00zled, 2022-06-15:
Is there still a possibility of getting BBLV apk for Android use? That would be huge as it would completely remove the need of a computer from the equation for everything BetaFlight for all but the most troublesome and/or involved configuration/analysis. I’m no dev, but from the outside, seems like it wouldn’t be terribly difficult, but I could easily be very wrong. Just trying to keep the hope alive. There is interest.

haslinghuis, 2024-05-13:
We have moved to [PWA](https://blackbox.betaflight.com)

closed 2024-05-13 by haslinghuis