betaflight/blackbox-log-viewer #492: BBE 3.6.0 video export double-dialog

meoso, 2021-01-09:
* When exporting video, the expected save-as dialog appears. Video export processes, but file is not populated during this time, then upon export-completion, a second dialog appears and the file successfully saves from this second dialog.
* Unknown if this is limited to Linux or exists in all OS's.
* Ran this on Ubuntu 20.04, both BBE v20210105.1590 & BBE v20201226.1471, did not digress to old version to find which one last worked.
* reinstall nor reboot fixes (just in case)
* `Linux 5.4.0-59-generic #65-Ubuntu SMP Thu Dec 10 12:01:51 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
* ScreenRecording of issue https://youtu.be/du9uUd3J3NM

thank you

mikeller, 2021-01-09:
Tested with Ubuntu 20.04, the double dialog reproduces.

McGiverGim, 2021-06-03:
I've made a PR as temporary fix for that. I will let this open because the underline bug remains, to see if we can fix it in a future maintaining the FileWriter support.

meoso, 2022-01-12:
honestly, the best solution for me was revert NW.js to 0.48.4 which was the last version that worked as expected.

haslinghuis, 2022-01-13:
@meoso Yes we have a lot of work to do. Chrome API's are deprecated and we need to adapt. See https://github.com/betaflight/betaflight-configurator/pull/2636

McGiverGim, 2022-01-28:
This was totally fixed in https://github.com/betaflight/blackbox-log-viewer/pull/553 so I close this.

closed 2022-01-28 by McGiverGim