betaflight/blackbox-log-viewer #497: GPS speed missing in the dropdown list of available fields

N4V1G4T0R, 2021-02-08:
It seems, the GPS Speed is in the log, but it is not shown as a choosable field in the dropdown menu. It's also not there in the CSV export.

According to @mikeller it's written into the blackbox file by Betaflight: https://github.com/betaflight/betaflight/blob/2467b32f91fbaa1cfb13d95db246a3116cb33b68/src/main/blackbox/blackbox.c#L1022-L1047

Line 1041 "blackboxWriteUnsignedVB(gpsSol.groundSpeed);"

BlackBox Explorer v3.5.0

mikeller, 2021-02-09:
@N4V1G4T0R: Have you tried the blackbox-tools (https://github.com/betaflight/blackbox-tools)? I think they can export the GPS speed.

N4V1G4T0R, 2021-03-05:
So, it appears to be a matter for the bb explorer, the data is there.

N4V1G4T0R, 2021-03-05:
So it appears to be a matter for the explorer, the data is there.

closed 2021-03-05 by N4V1G4T0R