betaflight/blackbox-log-viewer #498: MacOS Big Sur Crashes

granolaFPV, 2021-02-15:
1. Describe your problem:
Blackbox Explorer 3.5.0 crashes frequently on a MacBook Pro M1 running Big Sur 11.1. Often, but not always, the application crashes immediately after opening. Other times the application runs as expected for some time, but then crashes after an interaction. Also, other weird behavior is observed — e.g. the timeline at the bottom of the viewer window won't render properly.

2. Include ways to reproduce the problem:
Install Blackbox Explorer 3.5.0 on an M1 Mac running Big Sur 11.1. Open app and observe frequent crashes immediately upon opening. If the application continues to run, observe seemingly random crashes during normal use.

3. Provide as much information as possible about your hardware and software, including:
- MacBook Pro M1 running Big Sur 11.1
- Blackbox Explorer 3.5.0

Happy to provide additional information that would be helpful.

I assume this issue might be related: https://github.com/betaflight/betaflight-configurator/issues/2274

mikeller, 2021-02-16:
@granolaFPV: Can you please test with the latest build from https://github.com/betaflight/blackbox-log-viewer-nightlies/rele to see if this problem is fixed? If so the fix (an update of the NW.js framework) will be included in the next release, use a development build on the affected machine for now.

granolaFPV, 2021-02-18:
Thanks @mikeller for the nightly link. So far it's working great, as is the 10.8 nightly I've been running. I'll close the issue as I assume the NW.js fix is the root cause.

closed 2021-02-18 by granolaFPV