betaflight/blackbox-log-viewer #500: No video only sound

SebastianKumor, 2021-03-03:
Opening gopro MPEG-4 file shows black screen and when I play the video only sound is working

I used gopro hero 7 with latest software to record this video  in 4k 4:3 30 fps

I am running mac OS Big Sur 

I tried both latest release 3.5.0 and also pulled and built the master branch and same problem

I tried .mov file from my fatshark googles and that works fine.

I am thinking if I am missing some codec or something that could cause this issue?

McGiverGim, 2021-03-04:
It does not use the codecs of your system, it uses the ffmpeg library from here: https://github.com/iteufel/nwjs-ffmpeg-prebuilt

Maybe it does not support the gopro video codec?

SebastianKumor, 2021-03-04:
@McGiverGim could be, i tried both videos from hero 6 and 7 and same issue. I will also try to run it on windows if there is any difference

closed 2021-04-12 by github-actions[bot]

expipiplus1, 2021-04-25:
I have also encountered this on Linux, with several file encodings.

There are some repeated lines in the log along the lines of 

```
Errors:
Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Dec 16 2020. Please check build information of source that generated the binary.
[292425:292425:0425/104618.250344:ERROR:shared_context_state.cc(71)] Skia shader compilation error
```

However these do also appear on program load, before a video file has been selected.

I'm running `nw ./dist` after running `gulp dist`