betaflight/blackbox-log-viewer #501: Program crashed while export to csv log file

hyperion11, 2021-03-09:
link to log file
https://1drv.ms/u/s!AuFPVI_eDaZbhY1C00z8whcSC4KQIQ?e=BjV6gV
try stable 3.5.0 and latest nightly. Same problem - program crash.
FC Omnibus F4, fw Betaflight 4.2.5

McGiverGim, 2021-03-11:
I have tested a 3.5 version and it has worked. I've tested latest master and it has crashed. 

Maybe some memory problem, your file generates a 244Mb CSV file that we keep in memory to download.

I will test if I can find something that makes it less stable.

McGiverGim, 2021-03-11:
I can confirm... my 3.5 version worked because it was a custom 64 bit version build.

Export to CSV your file takes more than 3Gb of memory. I optimized a little the code some time ago, but is clear that it needs more work on it. The 64bit of the Blackbox can work with it, the 32bit not and crashes.

On the other side, @mikeller maybe is time to publish a x64 version. I doubt there is too much people using 32bit Windows versions now. We don't support Windows XP or Windows 7 anymore... What do you think?

hyperion11, 2021-03-12:
64bit windows version will be good solution i think.

mikeller, 2021-03-22:
@McGiverGim:

> On the other side, @mikeller maybe is time to publish a x64 version.

Yeah, I think we can switch to publishing the 64 bit version for Windows.

McGiverGim, 2021-03-22:
@hyperion11 I have pushed a PR to migrate to win64. Once merged the nightlies versions of the Blackbox Log Viewer will be 64bit versions.

Until then, if you want to test it I think you can download it from [here](https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=2266&view=artifacts&pathAsName=false&type=publishedArtifacts) is where Azure has published the artifact, but will only be available for some time.

closed 2021-03-28 by mikeller

haslinghuis, 2021-05-14:
For those still using `32bit` Windows use this instruction to build it yourself:

1. Install the `Nodejs` from here: https://nodejs.org/dist/latest-v10.x/node-v10.24.1-x86.msi
2. Clone the betaflight configurator (see the project for instructions, can also copy and expand the `zip` if not using `git`)
3. Step into the project directory and use

```
yarn install
yarn release --win32
```
Output will speak for itself.