betaflight/blackbox-log-viewer #513: Does not show 1080p video only sound and black screen

tn2009, 2021-05-28:
In the blackbox explorer, only 2.7 k, 4k videos are displayed. I add 1080p video at any fps, the video does not appear only sound. I checked on the video from the GoPro Hero 7 Black, Runcam Hybrid.

McGiverGim, 2021-06-02:
Are you sure all your videos use the same codecs? I've several 1080p videos, from a Runcam Split, and they work perfectly.

We use the ffmpeg library from here: https://github.com/iteufel/nwjs-ffmpeg-prebuilt so we only support the codecs that this library support.

tn2009, 2021-06-02:
Over the weekend I recorded a lot of videos via Rucam Hybrid. 1080p 60fps no video is displayed (black screen), only audio.

https://disk.yandex.ru/d/IurHH2ldSSbt1Q - information about the video file

closed 2021-07-10 by github-actions[bot]