betaflight/blackbox-log-viewer #519: Gyro debug lines don't overlap with with other lines of similar values

akonokhov, 2021-06-16:
When I'm putting gyro debug on the same graph with another fields that have the same or similar values, their lines do not overlap but have quite a big distance.
On the attached screenshot, according to the legend, all 5 lines should be somewhere close, at least those that show gyro values (723, 722, 722). But instead non-debug values are shown as expected and there is a distance between them and debug lines (Gyro 1 [roll], Gyro 2 [roll]).
It doesn't seem like a big issue, but it's quite confusing, because when there is a distance between lines, we expect them to have different values.
Blackbox log is attached.
Thanks.

[image]
[image]

[btfl_001.zip](https://github.com/betaflight/blackbox-log-viewer/files/6661629/btfl_001.zip)

akonokhov, 2021-06-16:
Forgot to mention that the behaviour is the same in v3.5.0 and in the nightly build (v3.6.0).

McGiverGim, 2021-06-16:
It seems that the debug mode `DUAL_GYRO_SCALED` that you are using has not been added to the Blackbox. I will add it.

McGiverGim, 2021-06-16:
@akonokhov you can test the change it if you want here: https://dev.azure.com/Betaflight/Betaflight%20Nightlies/_build/results?buildId=3056&view=artifacts&pathAsName=false&type=publishedArtifacts

akonokhov, 2021-06-30:
@McGiverGim , thanks!
I've tried that link but it doesn't show any artifacts. Also, tried to search Azure pipelines for that build, but didn't find anything.

McGiverGim, 2021-06-30:
The PR builds only last for some days... If you didn't catch it in time you will need to wait until this is merged or build it yourself.

closed 2021-09-03 by ctzsnooze