betaflight/blackbox-log-viewer #524: Analyzer zoom/scaling small screen cutoff

LekeFly, 2021-08-08:
[image]

I think the screenshot speaks for it self. Most of the information needed is here, but i would like see more to the left.

mikeller, 2021-08-10:
@LekeFly: No, it does not. Can you describe a bit better what change you would like to see?

LekeFly, 2021-08-10:
@mikeller My bad. Was supposed to write **to the right** my bad. And maybe the screenshot isn't the best either. 

[image]

There's more going on up in the right corner here, but i can't zoom/scale/scroll. I see on YouTube videos etc where they got higher resolution screens that it gives a better overview.

mikeller, 2021-08-10:
@LekeFly: So if I understand you right you'd like a way to zoom into the right side of the graph, instead of into the left side?

LekeFly, 2021-08-12:
[image]

Finally found an example. Yes like this @mikeller. I’m sorry for the unclear explanation

LekeFly, 2021-08-24:
Not sure why this says i closed the issue.

JoeriCapens, 2021-09-17:
It just looks like you need to log at a higher rate. To get results up to 1 kHz, you need to log at 2 kHz because of the Nyquist theorem.