betaflight/blackbox-log-viewer #533: Export CSV file

VerneZoe, 2021-09-21:
Hello developer
I found a problem: when the blackbox data exceeds 32 MB, the CSV file cannot be exported properly.
Can you fix it? Thank you.

McGiverGim, 2021-09-21:
What version are you using? The latest nightly in theory must fix this problems. If not please report again.

VerneZoe, 2021-09-21:
I'm using 3.6.0. I noticed this problem with version 3.5.0, and then I switch to the latest nightly version, and the problem persists.
The blackbox data source is OpenLager

McGiverGim, 2021-09-21:
Attach your log file to let others test it, to see if something can be fixed...

VerneZoe, 2021-09-21:
This blackbox data is used for gyroflow.Because it's a long range, the data is big.

https://1drv.ms/t/s!AvxPoj9XbsltnGRxC5Qhe4WhxGR7?e=T9cLMz