betaflight/blackbox-log-viewer #548: Newer builds of 3.6 Not Win7 Compatible

BrandonsBakedBeans, 2021-12-22:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
Win7 and ChromeOS on ARM is all I work with currently. I can't see the updated headers from 4.3 logs easily and the FFT charts don't display filters properly on the old build of 3.6 BBE I'm using. New nightly doesn't install on Win7.

haslinghuis, 2021-12-22:
Windows 7 is EOL for almost 2 years now. Please see: https://www.microsoft.com/en-us/windows/windows-7-end-of-life-support-information.

We now only offer 64 bit builds for Windows.

Please also see: https://github.com/betaflight/betaflight-configurator#windows-users

haslinghuis, 2022-01-23:
Does this install: https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20211029.4376

BrandonsBakedBeans, 2022-01-24:
> Does this install: https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20211029.4376

No it does not. I get an error with the description "This program does not support the version of Windows your computer is running"

If I have it installed on a win10 machine and copy the install directory it runs fine on win7-64bit

haslinghuis, 2022-01-24:
And this one: https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20210811.3500
(reverts NW.js 0.54.0 to 0.50.2, previous version was reverting to NW.js 0.54.0 from 0.54.1).

McGiverGim, 2022-01-28:
It is not a problem of the NW version, is a restriction we add to the installer. 

https://github.com/betaflight/blackbox-log-viewer/blob/5eed5dacfee687a1da5a9a7fa9bd94c0836f86f0/assets/windows/installer.iss#L65

This was done to be in line with the Configurator, because the NW version that we use is not compatible with Win7 and some features that the Configurator need, more info here: https://github.com/betaflight/betaflight-configurator/issues/2131

As a workaround, I think we can add the portable version like we add in the Configurator:
https://github.com/betaflight/betaflight-configurator/pull/2739
and you can use it in unsupported systems but without support for part of Betaflight.
https://github.com/betaflight/blackbox-log-viewer/blob/5eed5dacfee687a1da5a9a7fa9bd94c0836f86f0/assets/windows/installer.iss#L65

Another option is to remove the limitation in the installer, but I think is a good idea to have the same minimum requirements in all the Betaflight products.

PR #563 (2022-01-30): Add portable build to Windows
Follow up on: https://github.com/betaflight/betaflight-configurator/pull/2739

Fixes: #548
closed 2022-01-30 by haslinghuis