betaflight/blackbox-log-viewer #561: Crash on MacOS 11

bakwc, 2022-01-24:
### Describe the bug
Blackbox explorer crashes on MacOS

### To Reproduce
1) Run blackbox explorer

### Expected behavior
No crash

### Setup

- Blackbox Explorer version: 3.5.0
- MacOS 11.6.1

Core dump:

```
filipp@/Applications/Betaflight Blackbox Explorer.app/Contents/MacOS$ lldb nwjs
(lldb) target create "nwjs"
Current executable set to '/Applications/Betaflight Blackbox Explorer.app/Contents/MacOS/nwjs' (x86_64).
(lldb) r
Process 15323 launched: '/Applications/Betaflight Blackbox Explorer.app/Contents/MacOS/nwjs' (x86_64)
2022-01-24 19:55:07.760876+0300 nwjs[15323:3843677] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=15323
2022-01-24 19:55:07.760985+0300 nwjs[15323:3843677] SecTaskCopyDebugDescription: nwjs[15323]/0#-1 LF=0
2022-01-24 19:55:07.839184+0300 nwjs[15323:3843677] [plugin] AddInstanceForFactory: No factory registered for id <CFUUID 0x100735a10> 30010C1C-93BF-11D8-8B5B-000A95AF9C6A
2022-01-24 19:55:08.029641+0300 nwjs Helper (GPU)[15330:3843780] MTLIOAccelDevice bad MetalPluginClassName property (null)
2022-01-24 19:55:08.030851+0300 nwjs Helper (GPU)[15330:3843780] +[MTLIOAccelDevice registerDevices]: Zero Metal services found
Process 15323 stopped
* thread #4, name = 'StackSamplingProfiler', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x00007fff2a68dbfa libunwind.dylib`libunwind::CFI_Parser<libunwind::LocalAddressSpace>::decodeFDE(libunwind::LocalAddressSpace&, unsigned long, libunwind::CFI_Parser<libunwind::LocalAddressSpace>::FDE_Info*, libunwind::CFI_Parser<libunwind::LocalAddressSpace>::CIE_Info*) + 30
libunwind.dylib`libunwind::CFI_Parser<libunwind::LocalAddressSpace>::decodeFDE:
->  0x7fff2a68dbfa <+30>: movl   (%rdi), %r12d
    0x7fff2a68dbfd <+33>: leaq   0x4(%rdi), %rbx
    0x7fff2a68dc01 <+37>: movl   $0xffffffff, %eax         ; imm = 0xFFFFFFFF
    0x7fff2a68dc06 <+42>: movq   %rbx, -0x30(%rbp)
Target 0: (nwjs) stopped.

```

haslinghuis, 2022-01-24:
Please try: https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/latest

bakwc, 2022-01-25:
This one works fine, no crashes. Thank you.

closed 2022-01-25 by bakwc