betaflight/blackbox-log-viewer #576: Window unresponsive after loading workspace or log file in Ubuntu 22.04

musdom, 2022-06-08:
### Describe the bug

I can open a bbl file and it appears, but I can't scroll through the log or press any other button. I can only close and reopen BBL viewer.

### To Reproduce

Install and run BBL viewer in Ubuntu 22.04, then load a bbl file.

### Expected behavior

Can use BBL viewer normally

### Betaflight Blackbox Explorer version

3.5.0

haslinghuis, 2022-06-08:
@musdom please try https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20220521.1

musdom, 2022-06-08:
> @musdom please try https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20220521.1

This works! Thanks!

closed 2022-06-08 by musdom