betaflight/blackbox-log-viewer #587: `master` (Linux) does not see filenames with capital filename extensions

meoso, 2022-06-24:
### Describe the bug

* `master`'s open-file dialog does not see files with capital letter extensions 
* e.g. `BBL`, `AVI`, etc do not list; However `bbl`, `avi` do list.
* `Linux 5.10.0-15-amd64 SMP Debian 5.10.120-1 (2022-06-09) x86_64 GNU/Linux` (did not test other OS's)
* PR #574 does not solve.
* the code is correctly using the `/i` ignore case parameter, therefore i suspect `NWjs` as the problem-source
```javascript
                isLog = files[i].name.match(/\.(BBL|TXT|CFL|BFL|LOG)$/i),
                isVideo = files[i].name.match(/\.(AVI|MOV|MP4|MPEG)$/i),
                isWorkspaces = files[i].name.match(/\.(JSON)$/i);
```

[image]

### To Reproduce

on Linux, use 3.6.0, open files

### Expected behavior

list compatible files no matter the extension letter-case (lowerCase / capitalCase)

### Betaflight Blackbox Explorer version

3.6.0

haslinghuis, 2022-06-24:
Works on Ubuntu 20.04 LTS

Try
```
                isLog = files[i].name.toLowerCase().match(/\.(BBL|TXT|CFL|BFL|LOG)$/),
                isVideo = files[i].name.toLowerCase().match(/\.(AVI|MOV|MP4|MPEG)$/),
                isWorkspaces = files[i].name.toLowerCase().match(/\.(JSON)$/);
```

meoso, 2022-06-25:
> Works on Ubuntu 20.04 LTS

Also confirmed `Ubuntu 20.04.4 LTS` working.  My Debian 11 obviously different :(  both are XFCE4, fwiw.

closed 2022-06-25 by meoso

meoso, 2022-07-19:
findings:
* NWjs 0.50.0 and prior, works
* NWjs 0.51.0 though 0.65.1 fails
* NWjs 0.66.0 and .1 works

Caveat, all the above only tested with node v16.15.1