betaflight/blackbox-log-viewer #588: release `3.6.0` (Linux) export-video does nothing.

meoso, 2022-06-24:
### Describe the bug

* Export video does not (Linux).
* suspect NWjs
* other references #492 , #516
* tried multiple .BBL
* `Linux 5.10.0-15-amd64 SMP Debian 5.10.120-1 (2022-06-09) x86_64 GNU/Linux`
* https://youtu.be/8P3df754LTA

### To Reproduce

* Run released v3.6.0 or `master` as of 2202-06-24
* export video

### Expected behavior

process and save video

### Betaflight Blackbox Explorer version

v3.6.0

haslinghuis, 2022-06-25:
Works on Ubuntu 20.04LTS

meoso, 2022-06-25:
> Works on Ubuntu 20.04LTS

confirmed `Ubuntu 20.04.4 LTS` works; may be a gtk2 vs gtk3 thing :(

closed 2022-06-25 by meoso

meoso, 2022-07-19:
findings:
* NWjs 0.55.0 and prior, works
* NWjs 0.56.0 and later, even latest fails.

Caveat, tested with node v16.15.1 on Debian 11 only.