betaflight/blackbox-log-viewer #602: Allow user to display logged GPS Data

ctzsnooze, 2022-10-15:
### Is your feature request related to a problem? Please describe

Currently, if `blackbox_disable_gps` is not set to `ON`, and this is the default, GPS data will be saved to the log.

The log header will show:

```
H Field G name:time,GPS_numSat,GPS_coord[0],GPS_coord[1],GPS_altitude,GPS_speed,GPS_ground_course
H Field G signed:0,0,1,1,0,0,0
H Field G predictor:10,0,7,7,0,0,0
H Field G encoding:1,1,0,0,1,1,1
```

Or something similar.

These fields are really useful when debugging GPS rescue, but also for knowing groundspeed and course for testing purposes, mapping, and so on.

The problem:  **It is impossible to retrieve the data from the log.**  
The decoder for these fields was never implemented.

### Describe the solution you'd like

It would be fantastic, and much appreciated, if someone could parse these fields, and provide a means to:
- graph
- export into CSV
- display raw values in the info page

Some notes exist in the existing blackbox code [here](https://github.com/betaflight/blackbox-log-viewer/blob/313a9d55b93e5ec9a3659d56605a3db33357159d/js/flightlog.js#L449-L453).  

I'm not sure if anyone is able to do this, but it would be absolutely great if we could.

### Describe alternatives you've considered

I've asked around, but so far no luck.

atlaware, 2022-11-15:
https://github.com/betaflight/betaflight/discussions/11990
Hi, very good if can be implemented gps data export/show on blackbox-log-viewer.
But with https://github.com/cleanflight/blackbox-tools you can already get gps data into a gpx file or into csv file. from At least with bf <=4.2.
Try simply with: blackbox_decode --merge-gps [LOG FILE]
But as described there: https://github.com/betaflight/betaflight/discussions/11990 I have problems/
doubts with bf 4.3.
Maybe this can help.

Thank you.

ctzsnooze, 2022-11-15:
More info, in case someone with JavaScript brains can help....

Thanks to @nerdCopter we have some [code that will log the values identified in the header](https://github.com/nerdCopter/blackbox-log-viewer/commit/8f42a13d87bbde8c68eed13cceb6a4252ba9c077).

A sample of one block of fields from that logging shows that the values are in the log; I annotated the interpretation
```
G frame[0] 244610141  // blackbox time
G frame[1] 12         // num sats
G frame[2] -337786509 // GPS_coord[0] (latitude), divide by 1,000,000 to return 33.7786509 degrees south
G frame[3] 1512741715 // GPS_coord[1] (longitude) divide by 10,000,000 to return 151.2741715 degrees
G frame[4] 59         // GPS_altitude in tenths of a metre, ie 5.9m
G frame[5] 77         // GPS_speed in cm/s ie 0.77m/s
G frame[6] 1178       // GPS_ground_course (angle) in tenths of a degree, ie this means 117.8 degrees
```

In the firmware side, blackbox.c around line 253 indicates what goes into the fields.

```
#ifdef USE_GPS
// GPS position/vel frame
static const blackboxConditionalFieldDefinition_t blackboxGpsGFields[] = {
    {"time",              -1, UNSIGNED, PREDICT(LAST_MAIN_FRAME_TIME), ENCODING(UNSIGNED_VB), CONDITION(NOT_LOGGING_EVERY_FRAME)},
    {"GPS_numSat",        -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    {"GPS_coord",          0, SIGNED,   PREDICT(HOME_COORD), ENCODING(SIGNED_VB),   CONDITION(ALWAYS)},
    {"GPS_coord",          1, SIGNED,   PREDICT(HOME_COORD), ENCODING(SIGNED_VB),   CONDITION(ALWAYS)},
    {"GPS_altitude",      -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    {"GPS_speed",         -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)},
    {"GPS_ground_course", -1, UNSIGNED, PREDICT(0),          ENCODING(UNSIGNED_VB), CONDITION(ALWAYS)}
};

// GPS home frame
static const blackboxSimpleFieldDefinition_t blackboxGpsHFields[] = {
    {"GPS_home",           0, SIGNED,   PREDICT(0),          ENCODING(SIGNED_VB)},
    {"GPS_home",           1, SIGNED,   PREDICT(0),          ENCODING(SIGNED_VB)}
};
#endif
```

And `writeGPSFrame()` around line 1028 is the function that writes the values to the log:
```
static void writeGPSFrame(timeUs_t currentTimeUs)
{
    blackboxWrite('G');

    /*
     * If we're logging every frame, then a GPS frame always appears just after a frame with the
     * currentTime timestamp in the log, so the reader can just use that timestamp for the GPS frame.
     *
     * If we're not logging every frame, we need to store the time of this GPS frame.
     */
    if (testBlackboxCondition(FLIGHT_LOG_FIELD_CONDITION_NOT_LOGGING_EVERY_FRAME)) {
        // Predict the time of the last frame in the main log
        blackboxWriteUnsignedVB(currentTimeUs - blackboxHistory[1]->time);
    }

    blackboxWriteUnsignedVB(gpsSol.numSat);
    blackboxWriteSignedVB(gpsSol.llh.lat - gpsHistory.GPS_home[GPS_LATITUDE]);
    blackboxWriteSignedVB(gpsSol.llh.lon - gpsHistory.GPS_home[GPS_LONGITUDE]);
    blackboxWriteUnsignedVB(gpsSol.llh.altCm / 10); // was originally designed to transport meters in int16, but +-3276.7m is a good compromise
    blackboxWriteUnsignedVB(gpsSol.groundSpeed);
    blackboxWriteUnsignedVB(gpsSol.groundCourse);

    gpsHistory.GPS_numSat = gpsSol.numSat;
    gpsHistory.GPS_coord[GPS_LATITUDE] = gpsSol.llh.lat;
    gpsHistory.GPS_coord[GPS_LONGITUDE] = gpsSol.llh.lon;
}
#endif
```

So... the values are there, they are not difficult to decode.

It would be really great if someone could display these values and let us save them as CSV, like the other data in the log.

nerdCopter, 2022-11-16:
i don't think my code is worthy of PR, but someone could easily pull the code in this fashion:
```
git checkout -b gps-checks master
git pull https://github.com/nerdCopter/blackbox-log-viewer 20221103_GPS_maybe
```

haslinghuis, 2022-11-16:
Please provide a BBL with GPS data.

bonchan, 2022-11-16:
[here]([https://wetransfer.com/downloads/51f6ff1317f3f619e6d1ed1cbce27e4b20221116235336/cfcd1e6474a3cde48d2aaf3173fcaa9a20221116235349/26f39a](https://wetransfer.com/downloads/51f6ff1317f3f619e6d1ed1cbce27e4b20221116235336/cfcd1e6474a3cde48d2aaf3173fcaa9a20221116235349/26f39a?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid))

bonchan, 2022-11-16:
[here](https://wetransfer.com/downloads/51f6ff1317f3f619e6d1ed1cbce27e4b20221116235336/cfcd1e6474a3cde48d2aaf3173fcaa9a20221116235349/26f39a?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

nerdCopter, 2022-11-17:
another, provided by @ctzsnooze 
[GpsBaroIntegration1.BBL.zip](https://github.com/betaflight/blackbox-log-viewer/files/10031993/GpsBaroIntegration1.BBL.zip)

LinusThorsell, 2022-11-19:
What is the status of this feature?
Any PR's incoming or work done?

Personally looking to contribute if needed.

haslinghuis, 2022-11-19:
Didn't get further as:

[diff.txt](https://github.com/betaflight/blackbox-log-viewer/files/10047861/diff.txt)

bonchan, 2022-11-21:
I continued with @haslinghuis code and I got stuck trying to get the data to the chunk. 

this is the [commit](https://github.com/bonchan/blackbox-log-viewer/commit/e48ef6f4d9c80377f286d2455cf5472c164d3a8e), 
it has some comments to explain my problem.  

could anyone add the gps values to the chunk?

haslinghuis, 2022-11-24:
Fixed in #606

closed 2022-11-24 by haslinghuis