betaflight/blackbox-log-viewer #611: BBE 3.7.0 Nightly does not decode BaroAlt

aowi7280, 2022-11-28:
### Describe the bug

BB Nightly 3.7.0 (v20221126.30) does not decode the baro altitude in the View Log Header. Released version 3.6.0 works OK.

[image]

### To Reproduce

Install 3.7.0, open blackbox log, click the view log header icon on top row.

### Expected behavior

It should decode the altitude from the raw value. And doing it in either feet or meters would be nice.

### Betaflight Blackbox Explorer version

3.7.0

haslinghuis, 2022-12-02:
@aowi7280 please try with latest firmware as the parameter was renamed.

closed 2022-12-02 by haslinghuis

aowi7280, 2022-12-03:
It does not seem to work using nightly https://github.com/betaflight/blackbox-log-viewer-nightlies/releases/tag/v20221202.34