betaflight/blackbox-log-viewer #624: Can't run more than 3 windows и at the same time.

ekap12, 2023-01-18:
### Describe the bug

betaflight-blackbox-explorer_3.7.0_win64-portable  v20220919.20 you can open as many program windows as you want
betaflight-blackbox-explorer_3.7.0_win64-portable  v20220921.21 as well as all subsequent versions, only 2 program windows open at most. Windows 7. I know that this version of the operating system is not supported, but the same bug is possible in newer versions.

### To Reproduce

Run 3 times betaflight-blackbox-explorer_3.7.0_win64-portable  v20220921.21 and you have only 2 opened windows. doesn't show any errors. 

### Expected behavior

open as many windows as I need

### Betaflight Blackbox Explorer version

20220921.21 and all next

### Add any other context about the problem that you think might be relevant here

I tried to take the old file ffmpeg.dll and move it to a newer version of the program but it didn't help

nerdCopter, 2023-01-18:
Honestly, it's supposed to open only once, unless you click "New Window" in the BBE.

ekap12, 2023-01-18:
Usually I open the BBE by launching the log file in Windows Explorer or, for example, by clicking on the log file in the telegram chat. The option you suggested with the new window button partially solves my problem and allows you to open more windows, although this is still a bit inconvenient.

LordGG, 2024-02-02:
Hi, same issue here with 3.7 (win64-installer) and Windows 11, I can only open one instance of BBE when double-clicking Blackbox file or exe.

I know there is a button "New Window", but it was much more handy to be able to directly open multiple blackboxes by double-clicking on them :)

haslinghuis, 2024-02-02:
On nightly build:

[image]

haslinghuis, 2024-02-02:
No problem opening 5 instances on linux

[image]

LordGG, 2024-02-02:
I have reinstalled the latest nighty build and I am still unable to open multiple instances (without using "New Window" button), with Windows 11.

I have also resinstalled version 3.6 to rule out Windows 11 (untill recently I was using W10). No issue with 3.6, I can open as many instances as I want (at least 6).

haslinghuis, 2024-02-03:
Have used the new window button. Not sure why we should open multiple instance of the application.

LordGG, 2024-02-03:
No problem using the "New Window".

As @ekap12 wrote, it is more convenient to directly open a blackbox from the browser (forum, FB, discord, mail...) by just clicking the file, instead of clicking "new Window", browse to the downloaded file and open it. Same when opening multiple BB from the same location. The hassle won't kill me, but since I was used to that by just one click... I got lazy :)

haslinghuis, 2024-02-03:
Yes the no error reported would make it hard to reproduce. But I'm lazy to install - as I'm living on master all the time. Have to install it to use file association to check if anything raises the issue. Seems to work for first child not the second.

ekap12, 2024-02-03:
> it hard to reproduce

I think the easiest way to understand where the problem is that prevents you from running more than 3 copies of the program is to look at the difference between 2 builds. I think there’s not a lot of new code in 2 days, but that’s where the problem lies.

betaflight-blackbox-explorer_3.7.0_win64-portable v20220919.20
betaflight-blackbox-explorer_3.7.0_win64-portable v20220921.21

haslinghuis, 2024-02-03:
Note: http://docs.nwjs.io/en/latest/References/Manifest%20Format/#single-instance

nerdCopter, 2024-02-05:
i always wondered why command-line `--new-window` never worked. maybe this is fixable.

haslinghuis, 2024-02-05:
Workaround is to open a new window in BBL and drag and drop the BBL file on it.

haslinghuis, 2024-05-29:
Closing as we no longer maintain NWjs versions as we moved on to PWA version.

closed 2024-05-29 by haslinghuis