betaflight/blackbox-log-viewer #646: Blackbox Explorer can't open log, but log file looks correct.

z00mantwo, 2023-07-19:
### Describe the bug

Blackbox Explorer can't open log, but log file looks correct.

### To Reproduce

File created by OpenLager micro SD board. 

### Expected behavior

Able to open log file. 

### Betaflight Blackbox Explorer version

3.6.0

### Add any other context about the problem that you think might be relevant here

Here's a link to download the log file.

https://www.dropbox.com/scl/fi/k4t0z05owvgnh5aftgsih/LOG001.TXT?rlkey=n2okaljz7322zi0htkk57i9tq&dl=0

z00mantwo, 2023-07-19:
Start of file seems to have all the "H" lines. Then data data seems to become corrupt then there is a "End of log" at the end. Blackbox Explorer reports "Error opening log: No logs in this file could be parsed successfully" when trying to open it.

nerdCopter, 2023-07-21:
i could not open the file with old, new, and even cleanflight viewers.  file is somehow corrupt.

z00mantwo, 2023-07-21:
Thanks for checking. OpenLager board must be bad.

closed 2023-07-22 by haslinghuis