betaflight/blackbox-log-viewer #667: Add RPM in BBE data panel of eRPM (like previous with RPM)

SupaflyFPV, 2023-11-20:
### Is your feature request related to a problem? Please describe

Previous RPM data in BBE had the RPM number in the info panel - can we add this for eRPM also?

[image]

### Describe the solution you'd like

RPM number

### Describe alternatives you've considered

Using old RPM

nerdCopter, 2023-11-21:
@tbolin

tbolin, 2023-11-21:
@SupaflyFPV  This should already be implemented with https://github.com/betaflight/blackbox-log-viewer/pull/640 and https://github.com/betaflight/blackbox-log-viewer/pull/666.
Recently the name of the eRPM field was changed from `eRPM/100` to just `eRPM`, in https://github.com/betaflight/betaflight/pull/13123.
I made the adjustment to black box explorer in https://github.com/betaflight/blackbox-log-viewer/pull/666. 
However, for some reason there haven't been any nightly releases since that PR was merged 2 weeks ago.

(Alos note: There are no BF release where the field is called `eRPM/100`, so I decided to scrap backwards compatibility. 
The eRPM data in logs from before the field was renamed will not be properly formatted when using black box explorer nightlies released after https://github.com/betaflight/blackbox-log-viewer/pull/666.)

SupaflyFPV, 2023-11-21:
> @SupaflyFPV This should already be implemented with #640 and #666. Recently the name of the eRPM field was changed from `eRPM/100` to just `eRPM`, in [betaflight/betaflight#13123](https://github.com/betaflight/betaflight/pull/13123). I made the adjustment to black box explorer in #666. However, for some reason there haven't been any nightly releases since that PR was merged 2 weeks ago.
> 
> (Alos note: There are no BF release where the field is called `eRPM/100`, so I decided to scrap backwards compatibility. The eRPM data in logs from before the field was renamed will not be properly formatted when using black box explorer nightlies released after #666.)

ah cool thanks for this...ok I guess I'll wait for an updated nightly...can we push it thru?

haslinghuis, 2023-11-21:
Tried to build but there is a problem with appdmg dependencies (ds-store -> macos-alias) which is needed for OSX.

[image]

[image]

haslinghuis, 2023-11-21:
@McGiverGim @asizon We need someone with OSX to update ???

[image]

haslinghuis, 2023-12-22:
CI builds has been fixed.

closed 2023-12-22 by haslinghuis