betaflight/blackbox-log-viewer #691: Can't open "Freq vs throttle" plots for some logs after #668

mituritsyn, 2024-01-23:
### Describe the bug

Analyzer display doesn't show "Freq vs throttle" data on some logs after #668 
affected fields: gyro, unfiltered_gyro, gyro_scaled maybe other

[image]

[1.txt](https://github.com/betaflight/blackbox-log-viewer/files/14025399/1.txt)

### To Reproduce

- Open provided log file. not sure if it is matter some fields disabled in BF blackbox tab( setpoint and rc commands are on place ).
- Switch to second log in top right menu.
- Add gyro | unfiltered_gyro |  gyro_scaled  and press on any axis. 
- Analyzer display will appear with no data (it is hard to see need to hover the cursor on bottom left area).

f12 console shows error:

`Cannot plot analyser TypeError: Cannot read properties of undefined (reading '0')      grapher.js:878 `

 For first log it works as expected

### Expected behavior

"Analyzer display" shows gyro data.

### Betaflight Blackbox Explorer version

3.7.0 (master)

mituritsyn, 2024-01-23:
@tbolin could you kindly check it?

nerdCopter, 2024-01-23:
- using `bfd5ab7e`
- reproducible only on `Freq vs Throttle`, reproducible on a personal Sept 2023 log

PR #693 (2024-01-23): Fixes #691 Prevent out of bounds access when ploting Throttle vs Frequency
Fixes #691
commit a11ed062 (2024-01-23): Fixes #691 Prevent out of bounds access when ploting Throttle vs Frequency (#693)

Prevent out of bounds access when ploting Throttle vs Frequency
closed 2024-01-23 by haslinghuis