betaflight/blackbox-log-viewer #694: Reset analyserZoomX/Y state

mituritsyn, 2024-01-24:
### Describe the bug

Opening new log resets analyserZoomY or analyserZoomX scroll but persist actual plot scale

[image]
[image]

### To Reproduce

- open Analyze plot from any log 
- scale to maximum
- open another file or select next log in multilog file
- notice that analyserZoomY analyserZoomX state reseted to default but plot scale still maxed

### Expected behavior

plot scale corresponds to analyserZoomY state

### Betaflight Blackbox Explorer version

3.7.0 master

mituritsyn, 2024-02-24:
up

nerdCopter, 2024-03-26:
resolved by https://github.com/betaflight/blackbox-log-viewer/pull/717

closed 2024-03-26 by nerdCopter