betaflight/blackbox-log-viewer #732: New Analyzer option for Freq. vs Motor RPM

Remleyfpv, 2024-04-23:
### Is your feature request related to a problem? Please describe

No not a problem just want to have the option to see frequency vs RPM.  This is mainly due to these new RPM filter upgrades we are getting. I want to see the behaviour of the RPM to noise.

### Describe the solution you'd like

A new option in the drop down menu of the analyzer display window: Freq vs RPM. this would show as the name suggest a heat-map of the frequency vs average RPM of the motors.

### Describe alternatives you've considered

None

haslinghuis, 2024-04-23:
Did you try our new online [blackbox log viewer](https://blackbox.betaflight.com/)

[image]

Remleyfpv, 2024-04-23:
@haslinghuis I didn't even know there was an online blackbox log viewer. But awesome to see this was already be implemented there. Will it also be implemented in the offline blackbox log viewer?  And bonus question: Why is the RPM shown in Hz and not just RPM this IMO seems a bit confusing?

nerdCopter, 2024-04-23:
@Remleyfpv , the Web-App will be  "installable", at least in Chrome.
[image]

The last of the standalone builds are here: https://github.com/betaflight/blackbox-log-viewer-nightlies