betaflight/blackbox-log-viewer #770: Error opening log: No logs in this file could be parsed successfully

Dextyy1, 2024-09-02:
### Describe the bug

Good morning,
I have an issue when importing my blackbox's logs. I get the message "Error opening log: No logs in this file could be parsed successfully".
It is the first time that I try using blackbox, but I seems to work properly since I have files with data in it. I am able to open them with PIDToolbox, but not with Blackbox explorer.  
I tried the current 3.7.0 version, and also the previous 3.6.0, without success. I also tried to do another flight, but this time by defining blackbox mode on my arm switch, to be sure that it is correctly saved before unplugging the battery. The files seems ok, but I can't import any of them.

Do you have any idea of what can cause this issue ? I add the files attached here.

[btfl_001.zip](https://github.com/user-attachments/files/16840540/btfl_001.zip)

Regards.

### To Reproduce

I don't really know how to reproduce this case. I follow the following steps :
Setting the blackbox on "Onboard flash", frequency 2kHz, mode "GYRO_SCALED", with all data included. My quad is on betaflight 4.5.0. I delete backlogs before each of my tests flights, and exporting the files right after each one by using "activate mass storage device mode".

### Expected behavior

I probably have something wrong with my files, but it seems that the data in them are pretty good. I have no idea of what can go wrong, because we don't have any setup on betaflight, and it seems pretty consistent because the longer the flight is, the bigger the file get. Maybe there is a problem with the betaflight / Blackbox explorer version, if someone have an explanation and a solution, I will be happy to get it !

### Betaflight Blackbox Explorer version

3.7.0 and also 3.6.0

### Add any other context about the problem that you think might be relevant here

It is the first time that I use blackbox, so I don't know how it works exactly and if I did any error.

Drone spec :
FC : Mamba H543 Wifi (Dual Gyro)
Firmware : 4.5.0 BTFL
Betaflight Target : DIAT/MAMBAH743_2022B(STM32H743)
4 motors in bi-directionnal DSHOT600, for 3D flight

nerdCopter, 2024-09-02:
your file: header is corrupt
[image]

Dextyy1, 2024-09-02:
Thanks for your answer !
Do you have any idea of what can cause this ? I can't find anything about this, maybe a list of most common problem could help, I'm lost...

Dextyy1, 2024-09-19:
I forgot to update this ticket. It happend that this issue was because I was writing too much data on blackbox. After removing some data (gps, magnetometer...), I was able to have fully working logs. I think that 2kHz logs with every details were to fast for the circuit board, I don't have any issue now. 
Thanks for your help anyway, I think I couldn't be able to find it without your help.