betaflight/blackbox-log-viewer #792: Panning on timeline panel broken when View Table is activated

X-Stuff, 2024-11-16:
### Describe the bug

UI stucks when trying to scroll through log timeline if Table View was activated

### To Reproduce

Chrome
Version 130.0.6723.117 (Official Build) (64-bit)

https://blackbox.betaflight.com/

1. Open any log file

2. Click and hold left mouse button, somewhere on the timeline panel and move mouse left-right (to scroll through timeline):
 [image]

3. Click on View Table
[image]

 4. Click and hold left mouse button, somewhere on the timeline panel again and try to move mouse left-right:

### Expected behavior

As in 3.6.0 you should be able to hold mouse button and move though timeline

### Betaflight Blackbox Explorer version

3.7.0

nerdCopter, 2024-11-18:
@demvlad , interested in this?
[image]

closed 2024-11-18 by haslinghuis

X-Stuff, 2024-11-18:
[image]

@haslinghuis 
Still reproducible on Deploy Preview:
 
https://github.com/betaflight/blackbox-log-viewer/pull/793#issuecomment-2483349615

haslinghuis, 2024-11-18:
Can you share the BBL for investigation?

X-Stuff, 2024-11-18:
Sure!
[LOG00006.zip](https://github.com/user-attachments/files/17805132/LOG00006.zip)

haslinghuis, 2024-11-18:
Code trips over null values:

[image]

Second attempt in https://github.com/betaflight/blackbox-log-viewer/pull/794/commits/2efaa6340277cbe4d4305ac45cb393019ebbdfbb