betaflight/blackbox-log-viewer #804: Build targets for V3.7.0

randoassuser, 2024-12-27:
### Is your feature request related to a problem? Please describe

In the latest version, 3.7.0, there is only the source code, no ready-to-install builds for Mac, Linux, Windows etc. Not everyone can use the online version (offline working procedures) and it is cumbersome to build the source code for target mac (at least we have not managed to do that yet) Help! :(

### Describe the solution you'd like

A provided .dmg build of the source or clear instructions on how to build from source for Mac on macOS 14 and 15.

### Describe alternatives you've considered

Using the old version that provides a ready build has been the only alternative, which sucks when trying to use it with 4.5 logs

haslinghuis, 2024-12-27:
The online app can be used offline too. See this icon on the taskbar (on the right)

[image]

Perhaps we integrate blackbox-log-viewer with configurator at some time.

closed 2025-02-04 by github-actions[bot]

haslinghuis, 2025-07-27:
https://github.com/betaflight/blackbox-log-viewer/blob/master/README.md#latest-stable-version

nerdCopter, 2025-07-28:
in effort to curb this issue, i updated the description from "tip" to important" and added "4.0.0" verbiage.

[image]

Sortam, 2025-09-14:
> in effort to curb this issue, i updated the description from "tip" to important" and added "4.0.0" verbiage.

Such update does not make clear that 3.7.0 cannot exist as usual application the same way as 4.0.0.
Actually, there is no way to use the **released** 3.7.0 at all, as I understand. 
The online version is 4.0.0, and the link to version 3.7.0 leads to the "Releases" section, where only source code can be downloaded. 
This is useless to the end user.
This is awful release management I would say.

nerdCopter, 2025-09-15:
i'm confused as why the releases assets were removed -- Maybe because they fail to read 4.6.0 properly. However you can still get old 3.7.0 nightly which might even be newer then the purged 3.7.0 release: https://github.com/betaflight/blackbox-log-viewer-nightlies/releases

demvlad, 2025-09-16:
>  [@demvlad](https://github.com/demvlad) any issues on your side ?

I work in  Chrome and Edge, there are no issues.
These sliders have interest html/css implementation, will check other browsers.