betaflight/blackbox-log-viewer #805: Error opening log ...

TheIsotopes, 2025-01-03:
### Describe the bug

since latest master the log viewer opens no logs anymore.
[image]

@haslinghuis pls can you check this.

### To Reproduce

using latest master

### Expected behavior

working log viewer

### Betaflight Blackbox Explorer version

latest master

nerdCopter, 2025-01-03:
master opened 3 logs successfully for me.

TheIsotopes, 2025-01-03:
I forgot that the logs are from Betaflight 4.6.0 (5dc5a8e97). 

[BTFL_BLACKBOX_LOG_EGODRIFT_20241218.TXT](https://github.com/user-attachments/files/18301303/BTFL_BLACKBOX_LOG_EGODRIFT_20241218.TXT)

haslinghuis, 2025-01-03:
@TheIsotopes please check #806. Note while using firmware 4.6 development please always use latest master builds.

TheIsotopes, 2025-01-03:
@haslinghuis ok, with this fix it works again. thx

PR #806 (2025-01-03): Fix type error
- fixes #805
- [#802](https://github.com/betaflight/blackbox-log-viewer/commit/20def918a53638cf2cf130bd3359bbb67efa095c) modified a read-only variable.
closed 2025-01-03 by haslinghuis