betaflight/blackbox-log-viewer #811: Black Box Website not showing logs

Weeperr, 2025-01-30:
### Describe the bug

For the past couple days whenever I load I blackbox log onto blackbox.betaflight.com, it doesn't show any data, just the wave form at the bottom of the screen. I've tried two different quads to make sure that they're not the problem. I had been uploading logs daily for the previous couple weeks with no issues. 

### To Reproduce

load a black box log onto the website 

### Expected behavior

data should appear after log is loaded

### Betaflight Blackbox Explorer version

the website version

haslinghuis, 2025-01-30:
Please try https://master.dev.blackbox.betaflight.com/

Weeperr, 2025-01-31:
Thanks, this works

nerdCopter, 2025-01-31:
@blckmn , can the web-app be put in sync with master again please?

demvlad, 2025-02-05:
Share broken log please.

Weeperr, 2025-02-05:
If this was to me, I deleted that log ages ago and have been using the have just been using the linked black box configurator since. Some of my traces look strange though , should I send a log?

nerdCopter, 2025-02-06:
> [...] should I send a log? [...]

please use Betaflight Discord for support: https://discord.betaflight.com/

closed 2025-03-16 by github-actions[bot]