betaflight/blackbox-log-viewer #816: Error opening log: No logs in this file could be parsed successfully

JAKAMI99, 2025-03-14:
### Describe the bug

I've flown my Speedybee F405 Stack for the first time and wanted to see all the things that happened.
Somehow I always get a "Error opening log: No logs in this file could be parsed successfully" when trying to open ANY of the .BFL files :/ 
I also attached my settings

[LOGS.zip](https://github.com/user-attachments/files/19254160/LOGS.zip)

[image]

### To Reproduce

-

### Expected behavior

-

### Betaflight Blackbox Explorer version

 4.0.0

haslinghuis, 2025-03-14:
Try master.dev.blackbox.betaflight.com

JAKAMI99, 2025-03-14:
Thanks for the fast reply but does not work :/

haslinghuis, 2025-03-14:
? Speedybee F405 Stack 

- which one as V3 has issues with blackbox and osd being shared on same bus.

closed 2025-04-21 by github-actions[bot]